Nickduino/Somfy_Remote #2: Alternate approach to sending 433MHz signal

devyte, 2016-06-15:
Hi,

I'm opening this as an issue, because I have no idea how to send a message to you :p Feel free to close right away!

As I mentioned in the Nodemcu firmware repo, I'm working on an alternate approach to control somfy rollers. In my case, I installed outdoor aluminum rollers on all window in my house for security reasons, due to several robberies in the area. The motors are manufactured by the same factory as for somfy, but the receivers and hand controllers are another brand, and use the standard 433.92MHz band.

I thought to control the rollers from my phone, probably along the same lines as you are thinking. Therefore, I started looking into the 433MHz communication, looking to hack a controller, or maybe get a transmitter or something, but then after further thought I dropped the idea very early on. 

You said that your approach would have the same security as the somfy controllers, and that is exactly what got me worried. My understanding (and please correct me if I'm wrong) is that there is virtually no security. Anyone with a 433MHz receiver could listen to the commands sent by your transmitters (either a hand controller or transmitter attached to an ESP), and then mimic the same train of pulses. In my case, it means that with less than USD$10 worth of hardware, and minimal software, that person could open all the rollers that I just installed in my house!

So, I figured that the 433MHz receivers and hand controllers have to be replaced.

Here's my approach:
Use the ESP with a tiny webserver onboard to pulse one of two gpio outputs for a predefined length of time.  Each of those gpio outputs is connected to a relay big enough to drive a motor direction line. The length of the pulse is programmable, to allow different settings for the different rollers (different heights of the windows).
For an initial approach, what I intend is to connect the ESP relays in parallel with the 433MHz receiver relays, so that either can activate the motor directions. Yes, there is the danger that if one of the two is activating the roller in one direction, the other could activate it in the opposite direction, which could potentially damage the motor, but this is just for testing purposes. Once I figure all is working as intended, I intend to drop the 433MHz receivers.

Of course, my approach is intrusive: I have to go and open up the rollers and mess with the cables to install the ESP. Also, it requires one ESP per roller. In my case that's 23. But once done, the house is protected at minimum by the security of my wifi (wpa2), which is pretty good, and with some networking knowledge that could be vastly improved.

Anyways, I just thought to tell you about this :)

Nickduino, 2016-06-16:
First, have you actually tried it with your blinds?
Somfy is currently switching to the io-homecontrol® protocol, to which none of the code is relevant. I assume it's safer, so if you already have the good stuff, don't over-engineer it ;-)

You probably understood it but, just so it is clear to everyone else who might read it:
"You said that your approach would have the same security as the somfy controllers"
It **is** Somfy's RTS protocol, not just an equivalent (Although it's vsky279 who said it, failing to mention he's based everything on my work here).

And yes they fucked up since you can decode a frame and use this information to control the blinds.
I believe the risk is limited. Yes anyone could potentially sniff your data and use it against you with 2€ worth of electronics, **BUT**:
- they would need to be smart enough to built the device;
- they would have to be physically near your home for hours, waiting for you to activate your blinds.

Whereas with your ESP8266 (a module I love) solution, anyone on the planet could mess with them.
So it's a better protection against one attack but many more people would be in a position to attack it...

Assuming you need some extra protection, I would consider the following:
- connect the blinds through [a programmable breaker timer](http://g02.a.alicdn.com/kf/HTB1nOluJVXXXXbqXFXXq6xXFXXXa/Free-Shipping-DIN-RAIL-DIGITAL-PROGRAMMABLE-TIMER-SWITCH-220VAC-16A.jpg) which will cut the power during the night and when you're at work;
- go a step further and have your home automation know when you're not home and turn the blind's power off (with relays or whatever). Or, at least build a Somfy _decoder_ and have an ESP8266 send your phone a notification when someone is sending signals while you're away.

My two cents...

devyte, 2016-06-16:
Very interesting insights! :)

First off, apologies, I thought this was vsky's repo. I should have realized from your name that it was not. Sorry!

No, I don't have the good stuff :) It doesn't seem to be available here yet. They do sell motors with integrated receivers here, but they're the exact same thing as the 433MHz protocol. The receiver is just integrated inside the motor.

About the power cutoff, I'm already considering that. All blinds are on a separate electrical circuit, which I intend to cutoff when not home.

I'm also adding magnetic sensors to all the blinds as anti-tamper measures. These are ok to use with 433MHz, because cloning one just activates the alarm, but doesn't give access.

> anyone on the planet could mess with them
What exactly do you mean? Do you mean somebody could access my private LAN externally, and then access the module's interface to open the blinds?

> they would have to be physically near your home for hours, waiting for you to activate your blinds
Apparently not a problem. I got robbed a couple of weeks ago, and the timing strongly suggests that somebody watched us leave, and knew the house was empty. The thief entered 1 hour exactly to the minute after I left, and I was gone for less than 2 hours total.

> they would need to be smart enough to built the device
True, but that would be all that's keeping a thief from access, and he wouldn't have to build it, just buy a 433MHz cloner, which aren't hard to find. Also, recent MOs here show thieves with strangely advanced skill sets, which is what has me worried...

The notification when opening is a very good idea, I hadn't thought of that. Thank you!!

Interesting discussion, I greatly appreciate it!

Nickduino, 2016-06-16:
> First off, apologies, I thought this was vsky's repo. I should have realized from your name that it was not. Sorry!
Don't be sorry. I think vsky should have mentioned me.

>About the power cutoff, I'm already considering that. All blinds are on a separate electrical circuit, which I intend to cutoff when not home.
Easiest short term solution, by far.

>What exactly do you mean? Do you mean somebody could access my private LAN externally, and then access the module's interface to open the blinds?
Yes. That's a tough task but many people could perform it, not just the ones camping in front of your lawn.

>Apparently not a problem. I got robbed a couple of weeks ago, and the timing strongly suggests that somebody watched us leave, and knew the house was empty. The thief entered 1 hour exactly to the minute after I left, and I was gone for less than 2 hours total.
Still, they have to wait for you tu move the blinds, and then for you to leave your place.

>True, but that would be all that's keeping a thief from access, and he wouldn't have to build it, just buy a 433MHz cloner, which aren't hard to find.
A Somfy cloner or a dumb cloner?
Because the protocol sucks, but not that much: it's a rolling code, so you can't send the same frame twice. The blinds won't move the second time.

>Interesting discussion, I greatly appreciate it!
Shared feeling.

closed 2016-10-28 by Nickduino