Nickduino/Somfy_Remote #3: About the transceiver

piejanssens, 2016-10-28:
Could you share some details on your schematic of the transceiver that you used?
You say that you use one from an existing remote, which remote exactly?
What kind of antenna design are you using? What's your reach?

Nickduino, 2016-10-28:
Hi,

I used a hacked remote for a few weeks during testing but I would advise on using a (more compact and cheaper) 433 MHz emitter. You'll have to switch the oscillator for a 433.42 MHz version.

I'm using a home-made helicoidal antenna. It's enough for the whole house.

dpiller, 2017-01-01:
Hi Nick! I am very happy that you made this solution for Somfy - I plan to integrate that to my own smart home solution (currently developlin) - originally I wanted to drywire a somfy remote, but yours is much better. 
I would have two questions:
- How should I choose a remote number to avoid accidentally interferencing with my current normal remotes? I mean I don't know which actual numbers using my current remotes. Maybe they are using 0x1, 0x2, whatever, but I don't know 
#define REMOTE 0x121300  - 
- Can you shot a photo of your antenna, giving antenna size (centimeter) too?

Many thanks

David

Nickduino, 2017-01-01:
Hi,

- I don't know what address range Somfy used. It's **highly** unlikely you'll chose one you had already so I didn't bother. If you wanted to make things perfects, you could write a decoding sketch to intercept the frames of your current remotes;
- the antenna is about 17 cm long (a quarter of the wave length at 433 MHz) and I spiraled it around a pen, so that it wouldn't take too much space. There is no math behind from me, the receptors are sensitive enough for it to work well.

If you're developing a whole smart home solution, you might have something beefier than a microcontroller to run it. You might want to have a look at Pi-Somfy:
https://github.com/Nickduino/Pi-Somfy

closed 2017-03-05 by Nickduino