Nickduino/Somfy_Remote #5: Use of standard hardware

AndiTheBest, 2017-03-04:
Hi,

did you think about using a CC1101 transceiver module instead of a self modified hardware?
Its the chip used in the Busware CUL (used to send somfy commands in fhem).

Regards

Nickduino, 2017-03-05:
I hadn't.

I don't see why you couldn't set it to the right frequency and send the frame with it, **providing you can control the timing reasonably well** (you can't just use ELECHOUSE_cc1101.SendData(frame,size)).

rorie1982, 2017-03-22:
Hi

So what do you mean by "I hacked a remote to send my signals"? Did you used a original Somfy remote and solder it to your Arduino board? Is your hacked remote able to send and receive?

Nickduino, 2017-03-22:
Yes, I hard-wired the arduino to the remote. I don't think you can receive data with it but you should be able to get them with any cheap 433 MHz receiver.

closed 2017-03-22 by Nickduino

rorie1982, 2017-03-22:
So with one wire? I cant find any example on google how to connect this, can you add or send a picture or schema?

Nickduino, 2017-03-22:
Two wires, you need ground and +3.3V (or +5V). I don't know your remote so I can't tell you where to solder them.

The easiest, by far, is to use a transmitter with the proper crystal.

rorie1982, 2017-03-22:
I have a Somfy Telis 1 RTS remote, if you got the same one can you send me a picture?

Nickduino, 2017-03-22:
I have a Simu HZ remote, they une a different layout and different microcontroller.

For a picture with a Somfy remote, have a look there: https://forum.arduino.cc/index.php?topic=208346.0