Nickduino/Somfy_Remote #7: DDRD and PORTD with ESP8266 (NodeMCU)

marthocoo, 2017-08-15:
Thanks for this sketch! I'm looking to modify it to allow for controlling my Somfy awning over MQTT. 

Problem: I'm using a NodeMCU ESP8266, and the ESP8266's GPIO registers are laid out differently than an Arduino's, so DDRD and PORTD do not work (and throw "DDRD/PORTD was not declared in this scope" error on compile). 

The pinMode() and digitalWrite() functions do work, however. Are these commands an adequate alternative to DDRD (pinMode()) and PORTD (digitalWrite()) for setting the OUTPUT mode and the HIGH/LOW output of PORT_TX?

Nickduino, 2017-08-15:
Of course it will work. I was a bit cocky in using direct port manipulation but the _digitalWrite_ should do the job just as well.

Cheers,
   Nick

closed 2017-08-15 by Nickduino

rgr101, 2017-11-18:
@marthoc:
Did you succeed in porting the code to ESP? 
I believe there are some challenges with the behavior of delayMicroseconds etc. on the ESP??
How did you overcome that? 

I'm considering an ESP approach in the future, too - to setting up a smart home somfy gateway - maybe also with MQTT (not decided so far, as my next step I first have to setting up and being successful with the original approach on Arduino with the original Nickduino sample sketch... :) )