Nickduino/Somfy_Remote #8: 'DDRD' was not declared in this scope

ANASOMARY, 2017-09-05:
am getting errors when I try compiling the code for NodeMCU on Arduino IDE, following are the errors

Somfy_Remote:159: error: 'PORTD' was not declared in this scope

     PORTD |= 1<<PORT_TX;

     ^

Somfy_Remote:166: error: 'PORTD' was not declared in this scope

   PORTD |= 1<<PORT_TX;

   ^

exit status 1
'DDRD' was not declared in this scope

Nickduino, 2017-09-06:
Hi,

I used direct ports manipulation because I thought timing was absolutely critical. It turns out the blinds are very much forgiving and don't require that.

The PORTD etc. are specific to the microcontroller and won't work with NodeMCU but you can replace them by the regular digitalWrite() function. Someone actually forked my repository and did just that.

closed 2017-09-06 by Nickduino

ANASOMARY, 2017-09-07:
Hello,
please excuse my "newbiness" , but how can we replace PORTD and DDRD with digitalWrite() ? 
so I can replace (DDRD |= 1<<PORT_TX;) and (PORTD &= !(1<<PORT_TX);)  with digitalWrite(PORT_TX) ??? I would really really appreciate it if you could put me in the right direction.

Nickduino, 2017-09-09:
digitalWrite() needs the pin **and** the state as arguments.
https://cdn.arduino.cc/reference/en/language/functions/digital-io/digitalwrite/

See the forks:
https://github.com/Nickduino/Somfy_Remote/network

ANASOMARY, 2017-09-10:
Thank you so much, it worked!

Nickduino, 2017-09-10:
Happy my code is useful to others!