Nickduino/Somfy_Remote #10: New approach with SX1278 LoRa ESP32 board ?

rgr101, 2017-11-17:
Hi all, 

after some frustrating experiments with the standard RF transmitters in terms of serious range issues in my house (yes, I have adjusted the frequency by soldering a 433.42 MHz SAW resonator!), I dropped my plan to build up my own Somfy Remote based on this approach. 

So, I'm planning another approach based on a SX1278 LoRa ESP32 board, hoping to finally overcoming any range issues. Such boards are being offered for ~ $12 including OLED Display
on Aliexpress etc.

So, I hope to build a real good working Somfy Remote with these ambitions:

- support of up more than 25 channels (99 would be a good number ;-) )
- direct control as a "normal" remote by attached pushbuttons (channel +/- select, up, my, down, prog)
- wifi control by IP commands for smart home enabling (should be easy to adopt to ANY Smarthome by supporting secure(!) IoT/Smarthome standard protocols)

I have already ordered some of these boards - they should arrive in Germany in 2-3 weeks.  

Because my limited time and software/hardware skills I want to ask if anyone is interested in teaming up for this project?

Nickduino, 2017-11-17:
Cool board you chose!

I'm surprised you don't get better range. Did you solder any kind of antenna? Tried another transmitter/resonator?

On my side, I've been very impressed by the sensitivity of the blinds (provided you emit at the right frequency).

rgr101, 2017-11-17:
Yes, after soldering the 433.42 resonator and adjusting the inductor on receiver side, I soldered a lambda/4 antenna (just a 17.3 cm piece of solid copper wire) each on sender and receiver and tried first with a simple virtual wire sketch (two arduinos, one as sender, one as receiver. The sender I left stationary on a central location of the house (at the same place where my tinkered Telis-4 remote is placed which today is reliably controlling all blinds of the house over 3 floors). And with the receiver I moved through the house to all blinds. The range was not too bad in principle, but there were serious blind spots even at short distances (most frustrating was that at a distance of e.g. 5 meters the connection was lost when I moved myself into the direct line between sender and receiver - 

No, I didn't try another pair of sender and receiver or resonator so far - indeed I've got a package of 10 pairs from eBay - maybe I caught just chinese electronic scrap :), but I was so frustrated that I don't longer wanted to count on this technology, all the more after I discovered this superior board offering all needed components plus a display as a bonus :) ... and my original aim anyway was to do it with an ESP board for enabling Wifi controlling...

But seriously, there isn't a big community around this board so far so there will be kind of pioneer work to be done... and I'm not sure about a permission for the high rf power of the SX1278 (especially over here in Germany with our strong restrictions; on the other hands the transmission power can be software adjusted to the allowed 10 mW if needed......)

btw - it was quite hard to catch the 433.42 resonators and now I have 20 pcs left - so.......if anyone is interested ....?!??

Nickduino, 2017-11-18:
I don't get the way you used the code.

You just need the emitter! The receiver is already connected to your Somfy blinds!
(And yes, the chinese receivers using an inductor are shitty)

closed 2017-11-18 by Nickduino

rgr101, 2017-11-18:
I didn't use the code so far, I only made some independent RF send/receive tests as a first step for getting used to the "Arduino RF world"
So, I never came to the point yet to use the code together with the sender and a Somfy blind

But if you mean the receivers are shitty and the reason for the bad connections, I'll to a second approach with the Nickduino code - thanks for the hint!

Nickduino, 2017-11-18:
Oh yes, definitely try your emitter with the Somfy receivers: you'll be surprised by the range you'll achieve!

If you want to do RF projects for other purposes: try to use receivers with a SAW, not an inductor.

rgr101, 2017-11-24:
👍 👍 👍 👍 👍 It works!! And it works so smoothly - thank you so much for this great piece of code and encouraging me to retry it with this "simple" transmitter!! 
And you have been right: I'm REALLY amazed of the range: It works well through 2 steel concrete ceilings and 10 meters+ from one end of the house to the other. I didn't yet try all blinds but my confidence is back now!!

So, for me the next piece of work will be to building the "gateway" for a real smart home controlling of all of my 22 somfy motors... :) - I believe Gustaf's ESP approach is already very close to my vision, so I might have a deeper look in his fork now... 

[image]

Nickduino, 2017-11-25:
Glad you got it working!

Personnally, I'm using an untuned and compact helical antenna with no range concern.

For your other RF projects, a good receiver should look like this:
https://www.ebay.com/itm/1pcs-RXB6-433Mhz-Superheterodyne-Wireless-Receiver-Module-for-Arduino-ARM-AVR/401085388270?hash=item5d628d55ee:g:90UAAOSwr7ZW4BQZ

rgr101, 2017-11-25:
Yes, I'm glad, too...! I'll try the helical antenna, too, for getting the whole more compact... 
Ref. receiver: Ah great  - thx for your hint :)

mgoeller, 2020-02-18:
> So, for me the next piece of work will be to building the "gateway" for a real smart home controlling of all of my 22 somfy motors... :) - I believe Gustaf's ESP approach is already very close to my vision, so I might have a deeper look in his fork now...

Can you link me to Gustaf's ESP approach?