Nickduino/Somfy_Remote #11: Is it possible to send tilt commands?

buuuudzik, 2018-01-01:
Hello,

you did a very, very nice library;) I have an Arduino and now I am waiting for RF transmitter but when I checked your program I didn't see the tilt commands. Is there a way to do this via your library?

Nickduino, 2018-01-01:
Don't forget the 433.42 MHz SAW oscillator.

Tilt commands? Aren't you using up/down buttons for that? If not, it's certainly possible (and quite easy) to add this support but you'll need to know the command code. The only list I found is the following:

> 0x1 | My | Stop or move to favourite position
> 0x2 | Up | Move up
> 0x3 | My + Up | Set upper motor limit in initial programming mode
> 0x4 | Down | Move down
> 0x5 | My + Down | Set lower motor limit in initial programming mode
> 0x6 | Up + Down | Change motor limit and initial programming mode
> 0x8 | Prog | Used for (de-)registering remotes, see below
> 0x9 | Sun + Flag | Enable sun and wind detector (SUN and FLAG symbol on the Telis Soliris RC)
> 0xA | Flag | Disable sun detector (FLAG symbol on the Telis Soliris RC)

buuuudzik, 2018-01-01:
Thanks for fast respnse;)

When I will have the transmitter I will check then.

This implementation has such commands, but it looks that tilt plus(or upStep) use the up command but it must be little longer. For up is 100ms and for upStep is 300ms.
https://github.com/nodemcu/nodemcu-firmware/blob/master/lua_examples/somfy.lua

Also via original Somfy RS485 Transmitter there is a possibility to choose the value of tilting from 1 to 127.

Nickduino, 2018-01-02:
vsky279 having based his nodemcu code on my sketch, it shouldn't be too hard to implement the "tilt" command on your Arduino. Check out and give it a thought :
https://github.com/nodemcu/nodemcu-firmware/blob/master/app/modules/somfy.c

buuuudzik, 2018-01-06:
I'm still waiting for resonators but now I have the transmitter so I've checked by radio your code on Arduino UNO with some modification, now it is without webpage but only with API.

But it is little slow(~1s per command) so the best would be to use NodeJS and Orange PI because I currently have also some nice webpage/api server for Somfy and I can little change this to be appropriate for your library.

If you want I can also add this to your repository after finish.

But I need your help because I am not so good in C++ so if you could explain me how can I replace this code to be appropriate for Orange PI:
`DDRD |= 1<<PORT_TX; // Pin 5 an output
PORTD &= !(1<<PORT_TX); // Pin 5 LOW

void SendCommand(byte *frame, byte sync) {
  if(sync == 2) { // Only with the first frame.
  //Wake-up pulse & Silence
    PORTD |= 1<<PORT_TX;
    delayMicroseconds(9415);
    PORTD &= !(1<<PORT_TX);
    delayMicroseconds(89565);
}

PORTD ^= 1<<5;
PORTD |= (1<<PORT_TX);`

And another question where is PORTD declared because g++ produce an error?

buuuudzik, 2018-01-06:
Can replace in such way:
`DDRD |= 1<<PORT_TX; // Pin 5 an output
// to
wiringPiSetup();
pinMode(PORT_TX, OUTPUT);

PORTD &= !(1<<PORT_TX); // Pin 5 LOW
// to
digitalWrite(PORT_TX, 0);`

buuuudzik, 2018-01-06:
1) Can I replace:
`DDRD |= 1<<PORT_TX; // Pin 5 an output
// to
wiringPiSetup();
pinMode(PORT_TX, OUTPUT);

PORTD &= !(1<<PORT_TX); // Pin 5 LOW
// to
digitalWrite(PORT_TX, 0);`

closed 2018-01-06 by buuuudzik

Nickduino, 2018-01-06:
Yes.

I used direct port manipulation because I thought timing was of the utmost importance, turns out it's not and you can use _pinMode_ and _digitalWrite_.

buuuudzik, 2018-01-06:
Thank you very much for your help;)

This one is for me too hard:
`PORTD ^= 1<<5;`

What its purpose?

Nickduino, 2018-01-06:
It toggles the PORT.

`digitalWrite(yourPin, !digitalRead(yourPin));
should do the trick

buuuudzik, 2018-01-06:
Do you know how can I replace byte type from Arduino in c++? Now I have a "SEGMENTATION ERROR":(

Nickduino, 2018-01-06:
No idea what you're looking for, sorry.

buuuudzik, 2018-01-06:
No problem, thanks for all;)