Nickduino/Somfy_Remote #12: Hardware

raspdealer, 2018-01-23:
Hi Nickduino,

I will be more than interested to test your project with a raspberry pi.
Can you let me know which transmiter I can use? And how to set it up?

Feel free to contact me by email to chat in EN or FR, pseudo@gmail com

Thanks in advance

alxlaxv, 2018-01-23:
I am interested too.
I hope you can help here!

Nickduino, 2018-01-23:
Hello fellow frenchman,
alxlaxv,

My Python version (to be used with a Raspberry Pi) is available here: 
https://github.com/Nickduino/Pi-Somfy

We discussed the emitter here:
https://github.com/Nickduino/Pi-Somfy/issues/5
In short: any chinese crap from eBay will work but you'll need to change the oscillator.

closed 2018-01-23 by Nickduino