Nickduino/Somfy_Remote #14: Debugging

ido1990, 2018-05-01:
Hi!
I recently moved to apartment with somfy system.
I downloaded the code and changed it to fit the ESP8266 (I'm using Wemos D1 mini).
Did some soldering to the RF 433MHz module and replaced the crystal to 433.42MHz.
Now I'm trying to debug it why it's not working.
I probed the ESP8266 pin and I can see the signal but my somfy doesn't respond to the P command.

Any idea would be appreciated.
Thanks!

Nickduino, 2018-05-01:
Hi,

I'll try to help but I need to understand what you are doing.

1. What do you use to see the signal? An oscilloscope? A storage one? Can you see the frame and make sure it's the right one?
2. How did you try to use the P command?

ido1990, 2018-05-01:
Sorry, I forgot to mention.
1. I'm using an oscilloscope (DS1050Z).
I just saw the signal, didn't check it deep enough.

2. I opened the terminal console on the Arduino IDE then I pushed the program button on my somfy remote until it jogged up and down, then I typed P to the console and I saw the reply.

Nickduino, 2018-05-01:
0. Have you checked the ESP 8266 LUA version based on my job? https://nodemcu.readthedocs.io/en/master/en/modules/somfy/

1. Alright. If it was me, I'd check the frame is what it should be (at least that 0s are low and 1s are high) because that's fairly easy;

2. That's how it should work. So, I'd say check the wiring of the emitter. And check it some more.
    Do you have some kind of antenna attached? Are you close enough from the blind?
    100% sure crystal replacement is good?
    Have another emitter to test? If not, you could connect your actual remote emitter to your ESP8266.
    Possibly an Arduino?

Nickduino, 2018-05-01:
Also, gnaeser seems to have built an ESP8266-compatible version:
https://github.com/gnaeser/Somfy_Remote/commit/2369ea198a5458616f5174571479cef4531814b4

ido1990, 2018-05-01:
0. I don't really know how to use LUA so I didn't try.
1. I attached an image from the scope, showing part of the frame (high is 3.3v) from the data pin.
2. I tried 17cm wire as an antenna and also tried without.
The blinds are ~2 meters from the ESP8266.
I can't be sure 100% but I really want to try everything before I solder another crystal it wasn't easy doing it with soldering iron :)
I've another emitter but I'll need to solder again.
I think I'll try an arduino first.

Nickduino, 2018-05-01:
One important question I'm asking too late: it only works with the Somfy RTS protocol (99% of the time it is the RTS, but still...), not the io-homecontrol thing.
(https://www.somfy.fr/a-propos-de-somfy/technologies-et-compatibilites)

Well, if the rolling codes weren't incrementing with your ESP, there's something wrong with your code (EEPROM management, for instance).

I suggest you make my sketch work with the Arduino before switching to the ESP.

Long press on the program button of the actual remote, then send a "P" frame with the Arduino. If the rolling code increments every time, the code most likely is good. If the scope sees data, the wiring most likely is good. Range should not be an issue. So I'm inclined to think either your emitter or your oscillator are fucked.

Have you checked the solved issues? https://github.com/Nickduino/Somfy_Remote/issues?utf8=✓&q=is%3Aissue
Might be worth it to check the ones on the Python version as well (if it's a hardware problem)

ido1990, 2018-05-01:
Yes, I got the Telis 1 RTS Pure.
I'll try to replace the crystal and if it's helping, try different emitter. 

I'll read the solved issues again.

Nickduino, 2018-05-02:
That's one happy ending :-)