Nickduino/Somfy_Remote #17: Still don't get it  - How to control a Somfy Telis 4 RTS

mrohner, 2018-08-03:
How should I tell the software which of the four channels to use for the up down and stop commands?

Nickduino, 2018-08-06:
Not sure what you did.
If you want to control four blinds independantly, you'll need to simulate four different remotes (with four different addresses: line #31).

So, you either use four microcontrollers running four sketches with different addresses or only one with the ability of switching (what Somfy did with the Telis 4 RTS), in the latter case, you make the address a variable depending on the switch position.

Maybe I'll have some interesting inputs if you share more of your project with me.

mrohner, 2018-08-07:
Thanks for the feedback. 
I in fact simulate now three remotes (three addresses) in one program on one controller. Of course I will store 3 rolling codes in the EEPROM etc.
'''const unsigned int REMOTE[]={0x121300,0x121301,0x121302};// Balkony, Dining, Soleil blinds
unsigned int rollingCode[]={0,0,0};'''

I added also a blinds parameter to this subroutine to address the correct remote ..
'''void BuildFrame(byte *frame, byte button, byte blinds ) {
  unsigned int code[3];
  EEPROM.begin(32);
  EEPROM.get(EEPROM_ADDRESS, code);
  frame[0] = 0xA7; // Encryption key. Doesn't matter much
  frame[1] = button << 4;  // Which button did  you press? The 4 LSB will be the checksum
  frame[2] = code[blinds] >> 8;    // Rolling code (big endian)
  frame[3] = code[blinds];         // Rolling code
  frame[4] = REMOTE[blinds] >> 16; // Remote address
  frame[5] = REMOTE[blinds] >>  8; // Remote address
  frame[6] = REMOTE[blinds];       // Remote address'''

closed 2018-08-07 by mrohner