Nickduino/Somfy_Remote #20: Cloning the Remote Code without Reconfigure Somfy Motors

Hollako, 2018-10-03:
Hello,

First of all i want to thank you for this great hop to be able to make Somfy RTS work on ESP8266.

I will test it on the Wemos D1 mini but i would like to know from where I should take the HEX number of the Somfy RTS Remote, for example Telis 4 remote. the HEX number is located somewhere on the remote ?

Thank you,

Nickduino, 2018-10-03:
Thanks and...you're welcome!

Maybe you should have a look at the forks, some people did port it to the ESP8266.

As to cloning a remote, you won't be able to read the code anywhere on your remote, you'll have to code a sniffer to display it now that you know what the frame should look like.
**But** I don't get your point: it's much easier to create a new virtual remote and to register it (**in addition to the current one**) with your blind. Cloning the old remote would render it useless because you wouldn't be able to sync the two rolling codes.

closed 2018-10-03 by Nickduino

Hollako, 2018-10-03:
Got your point, noted !!

Thank you,