Nickduino/Somfy_Remote #23: Reliability Problem of RTS sending

Hollako, 2018-10-14:
Hello,

I am facing an issue of reliability with the 433.42 TX modules, I have 2 boards i replaced the 433.92 with the 433.42 MHz crystal.  Sometimes they work like charm and sometimes not. Am trying them on Arduino Uno since on ESP8266 is not working at all even if use the digitalWrite syntax.

Is the isuue may be related to the Antenna of the TX modules? and what is the preferable antenna to be added to the TX module for better coverage.

Noting that i modified the sketch to support up to 8 RTS remotes, i created many EEPROM addresses with 4 bytes reservation for each address and the RTS Remote addresses increment are 4 as well (i.e REMOTE1 address is 0x235440 the REMOTE2 is 0x235444)

Thank you.

Hollako, 2018-10-15:
Hello,

What you mean by the repetitions of the frame? i just duplicated the commands x 8 and created a cases where i switch between REMOTE addresses and EEPROM addresses.

Nickduino, 2018-10-15:
The sketch is sending the signal a few times. Maybe you should increase the number of repetitions, although it's not the most elegant solution

closed 2019-01-30 by Nickduino