Nickduino/Somfy_Remote #28: support more then 1 blind

4x4eb, 2020-10-02:
hi
I want to use one design to control several blinds.
how can I assign new commands like UP or DOWN for each blind individually?

Thanks
Eitan

Nickduino, 2020-10-02:
Either:

- Use Pi-Somfy: https://github.com/Nickduino/Pi-Somfy
- Add this functionality to the script: https://github.com/Nickduino/Somfy_Remote/issues/4

closed 2020-10-02 by Nickduino

4x4eb, 2020-10-02:
hi

thanks (-:

4x4eb, 2020-10-06:
another question please 

what is the reason for these lines?

if (EEPROM.get(EEPROM_ADDRESS, rollingCode) < newRollingCode) {
    EEPROM.put(EEPROM_ADDRESS, newRollingCode);

Nickduino, 2020-10-06:
The code sent by your remote change every time you use it (so that someone can't record it and play it back later), that's a rolling code. I store the current code in the EEPROM, so that you can power the Arduino off and not loose the sync.

4x4eb, 2020-10-07:
hi and thanks for the fast answer.

writing the last code to eeprom- I understand .
why do u compare it every power on to NEWROLLINGCODE ? its kind of constant as I understand.

Thanks for helping

Nickduino, 2021-10-10:
> writing the last code to eeprom- I understand . why do u compare it every power on to NEWROLLINGCODE ? its kind of constant as I understand.
> 

IIRC I wanted to be able to change the rolling code when uploading the sketch (if the const rolling code you set in the sketch is higher than the one in EEPROM, it will overwrite the latter with the former)