Nickduino/Somfy_Remote #29: Library Version

Legion2, 2020-12-03:
I created an Arduino Library based on this project https://github.com/Legion2/Somfy_Remote_Lib. It supports Arduino, ESP32 and  ESP8266. It can be downloaded in the Arduino IDE via the Library Manager.

Nickduino, 2020-12-04:
A library is a very good idea, thanks for having done this work.

In some circumstances, you may want it to run on a Raspberry Pi. Have you checked the Python version? https://github.com/Nickduino/Pi-Somfy

Legion2, 2020-12-04:
No, I'm using it with https://github.com/1technophile/OpenMQTTGateway and OpenHAB, so it is well integrated in my smart home setup.

Nickduino, 2020-12-04:
Perfect

closed 2020-12-04 by Nickduino

Legion2, 2020-12-05:
I also want to add that the CC1101 transceiver module is a very good alternative to self made 433.42 MHz modules. It is a drop in replacement for sending and receiving, the only drawback is that it requires a SPI connection to configure the frequency.

Nickduino, 2020-12-05:
Alright, I now  mention it in the README.md