Nickduino/Somfy_Remote #31: Provide a license

afoeder, 2021-03-21:
Hi @Nickduino, thanks a lot for your inspiring work that apparently is the baseline for many others :)

Would you please provide a license to your source code so that we can properly attribute you following the terms you've wished?

Thanks a lot!

Nickduino, 2021-03-21:
Glad you like it. Have you checked the (much more developed) Python version?

As for the licence, wouldn't that be enough?
[image]

afoeder, 2021-03-21:
ah, I apptly overlooked it looking for a LICENSE.md file or traces in the source code; thanks a lot!

I did check the python one but I specifically wanted to use an ESP so I went with this one.

Thanks again!

commit 225a0fb9 (2021-10-10): [TASK] Add LICENSE file

This adds the CC-BY-SA 4.0 license
information to a dedicated LICENSE
file following GitHub conventions.

Closes #31
closed 2021-03-23 by Nickduino