Nickduino/Somfy_Remote #33: Video tutorial

MakerMeik, 2021-07-18:
Hi Nickduino, I made a video tutorial on your awesome project. I hope you like it:
[https://youtu.be/9RhHrYqp9FU](url)

Nickduino, 2021-07-19:
Brilliant! I'll add it to the readme if you don't mind.

Just one thing: I think it would be useful to remind people in the title or description it only works with the RTS protocole, **not** with io-homecontrol

closed 2021-07-19 by Nickduino

MakerMeik, 2021-07-19:
Right, I overlooked that there is also the "io homecontrol" devices from Somfy. Otherwise I would have mentioned it in the video. I will definitely put a note in the Youtube video.
Do you know of a project that can also handle the io-protocol?

Nickduino, 2021-07-19:
> Do you know of a project that can also handle the io-protocol?

Nope, it's very closed. You'd need [such a module](https://www.conrad.nl/p/somfy-1822609-draadloze-module-2273656?WT.mc_id=affiliates%3Atradetracker%3A399611&utm_medium=affiliates&utm_source=tradetracker&utm_campaign=399611&utm_content=wouternieuwerth.nl) (one per shutter) or, better, a compatible hub (Connexoon) or home automation box (TaHoma)