Nickduino/Somfy_Remote #37: TUBE RTS Protocol // Somfy

filzek, 2022-12-02:
Hello Nick,

The TUBE has a RTS remote control but the remote control address always have 7 positions, so, beside the RTS default protocol it has add a initial 2 on it, so like 0x265822A using RFLINK to check to read the remote control it indentifies as RTS with 7 position in the ID but in RFLINK or in Somfy_Remote we are unable to send frames like this. Can you check it?

filzek, 2023-02-02:
we have found all changes to use with SOMFY RTS TUBE

frame[0] = 0x95; // is the COMMAND FRAME
frame[1] = (unsigned char)(15)<< 4; //is fixed always

so, the CMD as off:
#define TUBE_SCE_R1_CMD_PROG 0x9C
#define TUBE_SCE_R1_CMD_STOP 0x95
#define TUBE_SCE_R1_CMD_UP 0x96
#define TUBE_SCE_R1_CMD_DOWN 0x98
#define TUBE_SCE_R1_CMD_UP_DOWN 0x9A

Repeate the FRAMES to use special Command:
to make MY working just sent the STOP 10 times
to make the PROG to work just send it 25 times!

so all is done!!!

hzhh110, 2023-02-14:
> we have found all changes to use with SOMFY RTS TUBE
>
> frame[0] = 0x95; // is the COMMAND FRAME frame[1] = (unsigned char)(15)<< 4; //is fixed always
>
> so, the CMD as off: #define TUBE_SCE_R1_CMD_PROG [...]

Hello, can you provide complete information about TUBE? Thank you.

hzhh110, 2023-02-15:
> we have found all changes to use with SOMFY RTS TUBE
>
> frame[0] = 0x95; // is the COMMAND FRAME frame[1] = (unsigned char)(15)<< 4; //is fixed always
>
> so, the CMD as off: #define TUBE_SCE_R1_CMD_PROG [...]

TUBE FOR  frame[2] Don't know what logic

Nickduino, 2023-02-16:
> so all is done!!!

?
I didn't understand a thing you said but I guess you can fork the repo and adapt it to your setup

closed 2024-01-06 by Nickduino